UPB-PMRust/questions #1: Issue in following along the embassy-rs setup tutorial

UdrAlberto, 2024-03-16:
Good evening!
While following through the embassy-rs setup tutorial, after running the command `cargo install elf2uf2-rs`, I encountered the following error:

[image]

It was easily solved by installing a missing dependency, the package libudev, by running the command `sudo apt-get install -y  libudev-dev`, however I believe this should also be mentioned in the tutorial, in the Prerequisites section.

Thanks!

alexandruradovici, 2024-03-17:
Thank you for pointing out this. Could you please use the *Edit this page* function at tge bottom of the tutorial, edit the Linux section of the page by adding a info section (:::info ... :::) and send us a pull request?

UdrAlberto, 2024-03-17:
Done, thanks for the support!

closed 2024-03-17 by UdrAlberto