UPB-PMRust/questions #2: Can't find crate for 'test' error

12mihai05, 2024-03-18:
# Can't find crate for 'test' error

I'm encountering the error 'can't find crate for 'test'' in my Rust code from laboratory 2. After researching, I found two potential solutions. One suggests adding 'test = "0.1"' to dependencies in Cargo.toml, while the other suggests modifying settings in .vscode/settings.json. like this:
{
"rust-analyzer.cargo.target": "thumbv7em-none-eabihf",
"rust-analyzer.checkOnSave.allTargets": false
}
After applying the JSON solution, I encountered 271 other problems with missing crates, traits, and macros. Could you advise on how to solve this problem?

[image]

[image]

irina-b-dev, 2024-03-18:
Visual Studio Code might not load all the dependencies.
To check for building issues, please attach the output  `cargo build` command.

1. Check if you have `cargo` install : run `cargo --version`
2. Run `cargo build` in the _**bare_metal**_ directory

alexandruradovici, 2024-03-18:
The vscode settings seems to be the issue. This is the `settings.json` file from the lab https://github.com/UPB-FILS-MA/lab/blob/main/.vscode/settings.json.

12mihai05, 2024-03-19:
It worked after i changed the json file. Thank you.

closed 2024-03-19 by 12mihai05