UPB-PMRust/questions #3: Rust compilation error

sadtoni, 2024-03-22:
# Question

As discussed during the lecture, I encountered some problems while compiling the code. This time I am getting another error for the same code :)

Previous error:
```
error: could not compile `pac` (bin "pac") due to previous error
warning: build failed, waiting for other jobs to finish...
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "C:\\Users\\Antonio\\AppData\\Local\\Temp\\rustcujWwpl\\symbols.o" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\bare_metal-730169194aaa757d.2dm47ieyj8w3dmzm.rcgu.o" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\bare_metal-730169194aaa757d.2yxd05msshdugdtl.rcgu.o" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\bare_metal-730169194aaa757d.31d6u3yomygkbonv.rcgu.o" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\bare_metal-730169194aaa757d.39fbd0lv69pvwvcs.rcgu.o" 
"C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\bare_metal-730169194aaa757d.5ptchusznonwg3w.rcgu.o" "--as-needed" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\debug\\deps" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\bare_metal-d69b2f6feffcbcdb\\out" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-rt-cb475f9453fb37ea\\out" "-L" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "-Bstatic" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\librp2040_boot2-d7c7b48526434a6b.rlib" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\libembedded_hal-2c3de7b2ea0f18ad.rlib" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\libcortex_m_rt-b6c66d6fe5089c62.rlib" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\librustc_std_workspace_core-7df7ad8c02dab40f.rlib" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\libcore-5bc38c85d62440e3.rlib" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\libcompiler_builtins-8230ab960ae2a7aa.rlib" "-Bdynamic" "--eh-frame-hdr" "-z" "noexecstack" "-L" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "-o" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\bare_metal-730169194aaa757d" "--gc-sections" "-Tlink.x"
  = note: rust-lld: error: C:\Users\Antonio\Documents\GitHub\lab\target\thumbv6m-none-eabi\debug\build\cortex-m-rt-cb475f9453fb37ea\out\link.x:271: cannot find linker script device.x
          >>> INCLUDE device.x
          >>>         ^

error: could not compile `bare_metal` (bin "bare_metal") due to previous error
PS C:\Users\Antonio\Documents\GitHub\lab>
```

Current error:
```
PS C:\Users\Antonio\Documents\GitHub\lab> cargo build
   Compiling pio-proc v0.2.2
   Compiling input v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab\lab02\input)
   Compiling ex5 v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab\lab03\ex5)
   Compiling hal v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab\lab03\hal)
   Compiling bare_metal v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab\lab02\bare_metal)
   Compiling pac v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab\lab02\pac)
   Compiling embassy v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab\lab02\embassy)
error: linking with `rust-lld` failed: exit code: 1                                                                                                                                                      
  |
  = note: "rust-lld" "-flavor" "gnu" "C:\\Users\\Antonio\\AppData\\Local\\Temp\\rustcDS7JuP\\symbols.o" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\hal-aa9f2d77fa47a691.3al16cavota6p3vv.rcgu.o" "--as-needed" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\debug\\deps" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\hal-6866ec3326d0fc64\\out" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-dd9b77937a1e8551\\out" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-rt-cb475f9453fb37ea\\out" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\rp2040-pac-4c5e85e9511a36d2\\out" "-L" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "-Bstatic" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\libcortex_m_rt-b6c66d6fe5089c62.rlib" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\librustc_std_workspace_core-7df7ad8c02dab40f.rlib" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\libcore-5bc38c85d62440e3.rlib" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\libcompiler_builtins-8230ab960ae2a7aa.rlib" "-Bdynamic" "--eh-frame-hdr" "-z" "noexecstack" "-L" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "-o" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\hal-aa9f2d77fa47a691" "--gc-sections" "-Tlink.x"
  = note: rust-lld: error:
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Check that you actually use the device/hal/bsp crate in your code
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.

          rust-lld: error:
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Check that you actually use the device/hal/bsp crate in your code
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.

          rust-lld: error:
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Check that you actually use the device/hal/bsp crate in your code
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.

error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "C:\\Users\\Antonio\\AppData\\Local\\Temp\\rustcSN8DTD\\symbols.o" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\pac-5ce8788b29c8e696.21wnaikunk6zpqef.rcgu.o" "--as-needed" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\debug\\deps" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\pac-72e23b541205fdc4\\out" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-rt-cb475f9453fb37ea\\out" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\rp2040-pac-4c5e85e9511a36d2\\out" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-dd9b77937a1e8551\\out" "-L" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-n [... cut ...]
error: could not compile `bare_metal` (bin "bare_metal") due to previous error
PS C:\Users\Antonio\Documents\GitHub\lab> 
 ```
## Material read
<!-- place here any links to materials that are relevant to the question -->
I went to the lab repository, opened it with GitHub desktop and then opened a terminal in VS Code and tried "cargo build". I will attach a screenshot below of the new error.
![Screenshot (32)](https://github.com/UPB-FILS-MA/questions/assets/122456673/104d3f1f-9098-4ab0-95a4-8ee8bcdd3985)

alexandruradovici, 2024-03-23:
I updated your repository, please make sure to push all your code. The code that is now is missing some function and I get a different error.

alexandruradovici, 2024-03-23:
This is how your `Cargo.toml` in the `pac` folder should look like:

```toml
[package]
name = "pac"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
rp2040-boot2 = "0.3"
cortex-m-rt = { version = "0.7.3" }
cortex-m = { version ="0.7.7", features = ["critical-section-single-core"] }
rp2040-pac = { version = "0.6.0", features = ["critical-section"] }
embedded-hal = "1.0.0"
critical-section = "1.1.2"
```

alexandruradovici, 2024-03-23:
Seems to be a bug though.

sadtoni, 2024-03-24:
Tried to pull again the repository and to build again, I still get errors

```
PS C:\Users\Antonio\Documents\GitHub\lab> cargo build
   Compiling embassy-rp v0.1.0
   Compiling bare_metal v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab\lab02\bare_metal)
   Compiling ex5 v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab\lab03\ex5)
   Compiling pac v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab\lab02\pac)
   Compiling input v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab\lab02\input)
   Compiling hal v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab\lab03\hal)
   Compiling embassy v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab\lab02\embassy)
error: linking with `rust-lld` failed: exit code: 1                                                                                                                  
  |
  = note: "rust-lld" "-flavor" "gnu" "C:\\Users\\Antonio\\AppData\\Local\\Temp\\rustcSRofXD\\symbols.o" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\pac-5ce8788b29c8e696.21wnaikunk6zpqef.rcgu.o" "--as-needed" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\debug\\deps" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\pac-72e23b541205fdc4\\out" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-rt-cb475f9453fb37ea\\out" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\rp2040-pac-4c5e85e9511a36d2\\out" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-dd9b77937a1e8551\\out" "-L" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "-Bstatic" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\librp2040_pac-34686e2bc80ff7f7.rlib" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\libcritical_section-b433c2ba10048dbf.rlib" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\libcortex_m_rt-b6c66d6fe5089c62.rlib" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\libcortex_m-cf1a437e01633048.rlib" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\libembedded_hal-2fdeee70a6abe594.rlib" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\libvoid-19de1f7571267312.rlib" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\libnb-7e159f3bc672b2a9.rlib" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\libnb-5a3205537deea4aa.rlib" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\libvolatile_register-13c39b723fd6f459.rlib" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\libvcell-42acca2c2399a6c7.rlib" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\libbare_metal-0a8ec96b35650fe5.rlib" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\libembedded_hal-2c3de7b2ea0f18ad.rlib" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\librustc_std_workspace_core-7df7ad8c02dab40f.rlib" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\libcore-5bc38c85d62440e3.rlib" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\libcompiler_builtins-8230ab960ae2a7aa.rlib" "-Bdynamic" "--eh-frame-hdr" "-z" "noexecstack" "-L" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "-o" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\pac-5ce8788b29c8e696" "--gc-sections" "-Tlink.x"
  = note: rust-lld: error: undefined symbol: main
          >>> referenced by cortex_m_rt.285cc7b31f9833e1-cgu.0
          >>>               cortex_m_rt-b6c66d6fe5089c62.cortex_m_rt.285cc7b31f9833e1-cgu.0.rcgu.o:(Reset) in archive C:\Users\Antonio\Documents\GitHub\lab\target\thumbv6m-none-eabi\debug\deps\libcortex_m_rt-b6c66d6fe5089c62.rlib

error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "C:\\Users\\Antonio\\AppData\\Local\\Temp\\rustcreJhrP\\symbols.o" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\hal-aa9f2d77fa47a691.3al16cavota6p3vv.rcgu.o" "--as-needed" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\debug\\deps" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\hal-6866ec3326d0fc64\\out" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-dd9b77937a1e8551\\out" "-L" 
"C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-rt-cb475f9453fb37ea\\out" "-L" "C:\\Users\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\build\\rp2040-pac-4c5e85e9511a36d2\\out" "-L" "C:\\Users\\Antonio\\.rustup\\toolchains\\1.75-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "-Bstatic" "C: [... cut ...] sers\\Antonio\\Documents\\GitHub\\lab\\target\\thumbv6m-none-eabi\\debug\\deps\\bare_metal-730169194aaa757d" "--gc-sections" "-Tlink.x"
  = note: rust-lld: error: C:\Users\Antonio\Documents\GitHub\lab\target\thumbv6m-none-eabi\debug\build\cortex-m-rt-cb475f9453fb37ea\out\link.x:271: cannot find linker script device.x
          >>> INCLUDE device.x
          >>>         ^

error: could not compile `bare_metal` (bin "bare_metal") due to previous error
PS C:\Users\Antonio\Documents\GitHub\lab> 
```

alexandruradovici, 2024-03-24:
That is because you did not set the `#[entry]` macro on to of your `main` function.

alexandruradovici, 2024-03-24:
We updated the `Cargo.toml` for `pac`, take a look at https://github.com/UPB-FILS-MA/questions/issues/3#issuecomment-2016457609.

alexandruradovici, 2024-03-24:
Please push your code to github otherwise we cannot take a look at it.

alexandruradovici, 2024-03-27:
@sadtoni did it work?

sadtoni, 2024-03-27:
still no, this is the error i get

```
PS C:\Users\Antonio\Documents\GitHub\lab-solutions> cargo build
warning: output filename collision.
The bin target `solution` in package `embassy v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab02\embassy)` has the same output filename as the bin target `solution` in package `bare_metal v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab02\bare_metal)`.
Colliding filename is: C:\Users\Antonio\Documents\GitHub\lab-solutions\target\thumbv6m-none-eabi\debug\solution
The targets should have unique names.
Consider changing their names to be unique or compiling them separately.
This may become a hard error in the future; see <https://github.com/rust-lang/cargo/issues/6313>.
warning: output filename collision.
The bin target `solution` in package `embassy v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab02\embassy)` has the same output filename as the bin target `solution` in package `bare_metal v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab02\bare_metal)`.
Colliding filename is: C:\Users\Antonio\Documents\GitHub\lab-solutions\target\thumbv6m-none-eabi\debug\solution.dwp
The targets should have unique names.
Consider changing their names to be unique or compiling them separately.
This may become a hard error in the future; see <https://github.com/rust-lang/cargo/issues/6313>.
warning: output filename collision.
The bin target `solution` in package `ex5 v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab03\ex5)` has the same output filename as the bin target `solution` in package `embassy v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab02\embassy)`.
Colliding filename is: C:\Users\Antonio\Documents\GitHub\lab-solutions\target\thumbv6m-none-eabi\debug\solution
The targets should have unique names.
Consider changing their names to be unique or compiling them separately.
This may become a hard error in the future; see <https://github.com/rust-lang/cargo/issues/6313>.
warning: output filename collision.
The bin target `solution` in package `ex5 v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab03\ex5)` has the same output filename as the bin target `solution` in package `embassy v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab02\embassy)`.
Colliding filename is: C:\Users\Antonio\Documents\GitHub\lab-solutions\target\thumbv6m-none-eabi\debug\solution.dwp
The targets should have unique names.
Consider changing their names to be unique or compiling them separately.
This may become a hard error in the future; see <https://github.com/rust-lang/cargo/issues/6313>.
warning: output filename collision.
The bin target `solution` in package `hal v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab03\hal)` has the same output filename as the bin target `solution` in package `ex5 v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab03\ex5)`.
Colliding filename is: C:\Users\Antonio\Documents\GitHub\lab-solutions\target\thumbv6m-none-eabi\debug\solution
The targets should have unique names.
Consider changing their names to be unique or compiling them separately.
This may become a hard error in the future; see <https://github.com/rust-lang/cargo/issues/6313>.
warning: output filename collision.
The bin target `solution` in package `hal v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab03\hal)` has the same output filename as the bin target `solution` in package `ex5 v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab03\ex5)`.
Colliding filename is: C:\Users\Antonio\Documents\GitHub\lab-solutions\target\thumbv6m-none-eabi\debug\solution.dwp
The targets should have unique names.
Consider changing their names to be unique or compiling them separately.
This may become a hard error in the future; see <https://github.com/rust-lang/cargo/issues/6313>.
warning: output filename collision.
The bin target `solution` in package `input v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab02\input)` has the same output filename as the bin target `solution` in package `hal v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab03\hal)`.
Colliding filename is: C:\Users\Antonio\Documents\GitHub\lab-solutions\target\thumbv6m-none-eabi\debug\solution
The targets should have unique names.
Consider changing their names to be unique or compiling them separately.
This may become a hard error in the future; see <https://github.com/rust-lang/cargo/issues/6313>.
warning: output filename collision.
The bin target `solution` in package `input v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab02\input)` has the same output filename as the bin target `solution` in package `hal v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab03\hal)`.
Colliding filename is: C:\Users\Antonio\Documents\GitHub\lab-solutions\target\thumbv6m-none-eabi\debug\solution.dwp
The targets should have unique names.
Consider changing their names to be unique or compiling them separately.
This may become a hard error in the future; see <https://github.com/rust-lang/cargo/issues/6313>.
warning: output filename collision.
The bin target `solution` in package `pac v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab02\pac)` has the same output filename as the bin target `solution` in package `input v0.1.0 (C:\Users\Antonio\Documents\GitHub\lab-solutions\lab02\input)`.
Colliding filename is: C:\Users\Antonio\Documents\GitHub\lab-solutions\target\thumbv6m-none-eabi\debug\solution
[... cut ...]
   --> C:\Users\Antonio\.cargo\registry\src\index.crates.io-6f17d22bba15001f\critical-section-1.1.2\src\lib.rs:122:12
    |
122 | pub struct RestoreState(RawRestoreState);
    |            ^^^^^^^^^^^^

   Compiling gcd v2.3.0
Some errors have detailed explanations: E0308, E0428.
For more information about an error, try `rustc --explain E0308`.
error: could not compile `critical-section` (lib) due to 3 previous errors
warning: build failed, waiting for other jobs to finish...
PS C:\Users\Antonio\Documents\GitHub\lab-solutions> 
```

sadtoni, 2024-03-27:
the code is here, i tried the cargobuild
[https://github.com/sadtoni/lab-solutions](url)

sadtoni, 2024-03-27:
I proceeded like this and the issue was solved. I attached a screenshot of the command prompt with the success message.
Thank you!

[image]

closed 2024-03-27 by sadtoni