UPB-PMRust/questions #4: PICO BOARD

IoanaP0711, 2024-03-23:
As discussed during the lecture, here's the link of the board that I've told you about that I was thinking of using with a LEGO.

https://shop.pimoroni.com/products/inventor-2040-w?variant=40053063155795

DTeodor-Alexaandru, 2024-03-24:
What is the aim of the project? This board comes with relatively modest power and current constraints. It's important to handle the board with care to avoid any damage to the MCU and motor driver due to its restricted capacity.

alexandruradovici, 2024-03-24:
I think you will have to by the motors from them, I'm not sure if they are compatible with what Lego provides, and their website does not seem to give any documentation.

DTeodor-Alexaandru, 2024-03-24:
Please provide all the the necessary information and links for the hardware you intend to integrate.

alexandruradovici, 2024-05-11:
@IoanaP0711 is this still an issue?

closed 2025-03-14 by alexandruradovici