UPB-PMRust/questions #5: [lab03] Ex 5 #[embassy_executor::task]

NegrilaRares, 2024-03-27:
# Question
This solution for lab03 ex 5 does not seem to work as intended for me.

The led on pin PIN_0 blinks as intended but when the button on pin PIN_12 is pressed, both the led on PIN_0 and the led on PIN_25 change state.

Upon further testing the code would seem that the button not only does somehow interacts with PIN_0, but also seems to reset the values of any declared variables meant to limit the number of blinks from the led on PIN_0.

More specifically like this: 

let mut var: i32 = 0; 
loop {
        var = var +1;
        if var < 20
        {
        led.toggle();
        Timer::after_millis(200).await;
        }
    }

## Question

## Details
- Link in the lab: https://github.com/UPB-FILS-MA/lab-solutions/blob/main/lab03/ex5/src/bin/solution.rs
- Link to your code:

## Material read

alexandruradovici, 2024-03-27:
@TeodoraMiu can you test this example please?

closed 2024-03-28 by NegrilaRares

alexandruradovici, 2024-03-28:
What was the problem?

NegrilaRares, 2024-03-29:
It would seem that i had a misconception with the PIN values from peripherals. I wrongly used the pin number instead of the gpio index. 

I do not know how the pins turned on the right way with the wrong values but the button behavior was affected. 

After figuring out the right index for PIN everything worked.