UPB-PMRust/questions #8: Facing difficulties with the project documentantion

economuteo, 2024-04-05:
# Question

After a brief discussion with one of the representatives of the laboratory, Teodor Dicu, we clearly understood that the initial idea for our project was not enough and in order for the project to be considered good enough, we needed to implement a Kalman filter.  Now we are kind of confused and don't know where to start. We are scared to follow our own trajectory, searching all over the place for documentation, hoping to make all the components fit together and implement the Kalman filter. Could you please help us with a trajectory and some materials?

## Hardware used
LCD 1602 
Temperature sensor
Raspberry PI PICO HW

# Materials read

alexandruradovici, 2024-04-05:
@DTeodor-Alexaandru can you take this please?

DTeodor-Alexaandru, 2024-04-06:
Check out this well-written article on the Kalman Filter:
https://www.intechopen.com/chapters/76884

economuteo, 2024-05-11:
> @economuteo do you have any additional questions?

Yes! I have just a single question for now. What program should we use for the schematic in order for Teodor or Irina to print it on the 3D printer?

economuteo, 2024-05-11:
> > @economuteo do you have any additional questions?
> 
> Yes! I have just a single question for now. What software should we use for the schematic in order for Teodor or Irina to print it on the 3D printer?

We have already thought about how the model is going to look like. We just need the software.

alexandruradovici, 2024-05-11:
Any software that exports to STL files should be fine. I think we used [TinkerCad](https://www.tinkercad.com).

economuteo, 2024-05-11:
> Software that exports to STL files should be fine. I think we used [TinkerCad](https://www.tinkercad.com).

Ok! Thank you very much!