UPB-PMRust/questions #10: [project] Problem with multiple inputs in a single task.

NegrilaRares, 2024-04-10:
# Question

## Question
While trying to make a task with multiple inputs without the use of wait_for_falling_edge().await or wait_for_rising_edge().await like the following example:

```rust
#[embassy_executor::task]
async fn oct_input(
    channel_sender_octave: Sender<'static, ThreadModeRawMutex, OctLvl, 64>,
    mut sw_1:Input<'static, PIN_27>,
    mut sw_2:Input<'static, PIN_28>,
    mut sw_3:Input<'static, PIN_22>,
){
    loop{
        if sw_1.is_high(){
            channel_sender_octave.send(OctLvl::ONE).await;
        }
        if sw_2.is_high(){
            channel_sender_octave.send(OctLvl::TWO).await;
        }
        if sw_3.is_high(){
            channel_sender_octave.send(OctLvl::THREE).await;
        }
        Timer::after_millis(100).await;
    }
}
```
(The inputs are pull::down)

When i tried to run it the raspberry pi pico did not perform the task or any of the other tasks in my main

Thinking that using too many inputs was the problem I tried this but failed as well:

```rust
#[embassy_executor::task]
async fn oct_input_1(
    channel_sender_octave: Sender<'static, ThreadModeRawMutex, OctLvl, 64>,
    mut sw_1:Input<'static, PIN_27>,
){
    loop{
        if sw_1.is_high()
        {
            channel_sender_octave.send(OctLvl::ONE).await;
        }
        else {
            continue;
        }
        Timer::after_millis(100).await;
    }
}
```

The only version that worked is the following one but this will not work in the initial case with multiple inputs:

```rust
#[embassy_executor::task]
async fn oct_input_1(
    channel_sender_octave: Sender<'static, ThreadModeRawMutex, OctLvl, 64>,
    mut sw_1:Input<'static, PIN_27>,
){
    loop{
        sw_1.wait_for_rising_edge().await;
        channel_sender_octave.send(OctLvl::ONE).await;
    }
}
```

Any ideas as to how i could achieve the initial task?

## Details
- Link to your project repository: https://github.com/UPB-FILS-MA/project-NegrilaRares

## Material read

https://docs.rs/embassy-rp/latest/embassy_rp/gpio/struct.Input.html#method.is_high

TeodoraMiu, 2024-04-10:
The first two cases use the `is_high()` function, which isn't asynchronous. It just checks whether or not the button is high at a given moment, in your case, every 100 ms. If the button is not pressed down exactly when that condition is checked, it won't enter the `if` statement. It's not efficient to do it this way.

The `wait_for_rising_edge()` function, on the other hand, *is* asynchronous. This means that your `oct_input` function will resume execution once the `wait_for_rising_edge` future has completed.

In the case of waiting for multiple different button presses, you would have multiple `wait_for_rising_edge` futures. What you could do to achieve a similar behavior to what you presented in the third case is you could use `select`. `Select` waits until one of multiple futures completes, and then executes code depending on which future completed first. In other words, for whichever button was pressed first, that case is handled. 

You can find a more detailed explanation in [this section](https://embedded-rust-101.wyliodrin.com/docs/lab/05#awaiting-multiple-futures) of Lab 05.

NegrilaRares, 2024-04-10:
Thank you this was exactly what i was looking for.

Will leave this code here in case someone has a similar problem and wants to see the end result:

```rust

use embassy_futures::select::Either3::{First as First3, Second as Second3, Third as Third3};
use embassy_futures::select::select3;

#[embassy_executor::task]
async fn oct_input(
    channel_sender_octave: Sender<'static, ThreadModeRawMutex, OctLvl, 64>,
    mut sw_1:Input<'static, PIN_27>,
    mut sw_2:Input<'static, PIN_28>,
    mut sw_3:Input<'static, PIN_22>,
){
    loop{
        let select = select3(sw_1.wait_for_rising_edge(), sw_2.wait_for_rising_edge(), sw_3.wait_for_rising_edge()).await;

        match select {
            First3(switch_1) => {
                channel_sender_octave.send(OctLvl::ONE).await;
            }
            Second3(switch_2) => {
                channel_sender_octave.send(OctLvl::TWO).await;
            }
            Third3(switch_3) => {
                channel_sender_octave.send(OctLvl::THREE).await;
            }
        }
    }
}

```

NegrilaRares, 2024-04-10:
forgot to close issue with last comment

closed 2024-04-10 by NegrilaRares