UPB-PMRust/questions #14: [project] Issues with the usb logger

MarioMario7, 2024-05-07:
# Question
The embassy logger doesn't display anything in the console,but complies without error and i think i might have a problem with the usb port of the pico beacuse i've succesfully ran similar code at the lab. I opened this issue to check if the setup in my code is incorrect. I have attached the relevant code, the file is also available on my project github under the title `uart.rs`.
## Question

```rust
#![no_std]
#![no_main]

use cortex_m::asm;
use embassy_executor::Spawner;
use embassy_rp::uart;
use {defmt_rtt as _, panic_probe as _};
use embassy_rp::gpio::{Level, Output};

// Logger
use embassy_rp::bind_interrupts;
use embassy_time::{Duration, Timer};
use heapless::Vec; // Might use
use static_cell::StaticCell; // Might use
use log::{info, warn};

// USB driver
use embassy_rp::peripherals::USB;
use embassy_rp::usb::{Driver, InterruptHandler as USBInterruptHandler};

bind_interrupts!(struct Irqs {
    USBCTRL_IRQ => USBInterruptHandler<USB>;
});

#[embassy_executor::task]
async fn logger_task(driver: Driver<'static, USB>) {
    embassy_usb_logger::run!(1024, log::LevelFilter::Info, driver);
}

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let p = embassy_rp::init(Default::default());
    let config = uart::Config::default();

    // USB logger driver
    let driver = Driver::new(p.USB, Irqs);
    spawner.spawn(logger_task(driver)).unwrap();

    Timer::after_secs(2).await;

    info!("test-logger");

    Timer::after_nanos(20).await;

    let mut uart = uart::Uart::new_blocking(p.UART0, p.PIN_0, p.PIN_1, config);

    let mut buf = [0; 1];

    let mut trig = Output::new(p.PIN_4, Level::High);
    let mut trig_toggle_enabled = true; // Flag to control trig toggle

    loop {
        info!("test-logger");

        if trig_toggle_enabled {
            trig.toggle();
        }

        if buf.len() > 0 {
            trig_toggle_enabled = false; 
            buf = [0; 1];
        } else {
            trig_toggle_enabled = true;
        }

        // Read into buf
        if let Ok(_) = uart.blocking_read(&mut buf) {
            // Since read was successful, buf will contain the received data
        }
    }
}
```

## Details
- Link to your project repository: https://github.com/UPB-FILS-MA/project-MarioMario7

alexandruradovici, 2024-05-07:
@TeodoraMiu do you have time to take a look?

@MarioMario7 how are you running this?

MarioMario7, 2024-05-08:
> @MarioMario7 how are you running this?

thank you for the reply. due to my setup, i rename the file to main.rs to build it and then run the usual commands:

`cargo build --release --target thumbv6m-none-eabi`

and then

`elf2uf2-rs -d -s [PATH TO BINARY FILE]`

TeodoraMiu, 2024-05-08:
In the main loop, you should delay a bit before printing something over the serial. I tested your code and it works if you add a `Timer::after_secs()` call inside the loop.

MarioMario7, 2024-05-08:
> In the main loop, you should delay a bit before printing something over the serial. I tested your code and it works if you add a `Timer::after_secs()` call inside the loop.

thank you for the help, this does fix the problem

closed 2024-05-08 by MarioMario7