UPB-PMRust/questions #15: PWM_SLICE not working properly

cristianaprecup, 2024-05-10:
# Question

PWM_SLICE1 is configured for pin2 and pin3:
https://github.com/embassy-rs/embassy/blob/a7eb9bced1daa6e5f9bdf2c888d5920e33786b0b/embassy-rp/src/pwm.rs#L370-L371 ,
but cannot be used in the same time.

[image]

This is an issue because it reduces the number of pins available.

## Material read
https://github.com/embassy-rs/embassy/blob/a7eb9bced1daa6e5f9bdf2c888d5920e33786b0b/embassy-rp/src/pwm.rs

alexandruradovici, 2024-05-10:
The problem is that the `Pwm::new_output_a` function takes ownership of the `peripherals.PWM_SLICE1`. This is done to prevent using the same PWM Slice with two different `Pwm` drivers. Looking at the documentation of the `Pwm` driver, you can see a `Pwm::new_output_ab` function that takes two pins. The `PwmConfig` structure is only one, as it contains the `compare_a` and `compare_b` values. This means that it controls both of the channels.

cristianaprecup, 2024-05-10:
Now it's working as expected, thank you.

This is the implementation using `Pwm::new_output_ab` function:
```rs
let mut pwm_blue_green = Pwm::new_output_ab(peripherals.PWM_SLICE1, peripherals.PIN_2, peripherals.PIN_3, config_blue_green.clone());
```

mihaiprodann, 2024-05-10:
Hi,
As I see in the source file (https://github.com/embassy-rs/embassy/blob/a7eb9bced1daa6e5f9bdf2c888d5920e33786b0b/embassy-rp/src/pwm.rs#L370-L371), there are 4 pins that are using SLICE1, but I can't find any proper function that takes usage of 3 or all 4 of them.
Is there anything that I'm missing? Thank you.

alexandruradovici, 2024-05-10:
Each P slice has two channels and each channel can be mapped to one pin at a time.

mihaiprodann, 2024-05-10:
I'm not sure I completely understand this. Could you, please, show how this should be implemented in the case of my code?

alexandruradovici, 2024-05-10:
You cannot use four pins with ome PWM slice. You can use a maximum of two pins.

mihaiprodann, 2024-05-10:
It is clear now, thank you!

closed 2024-05-10 by cristianaprecup