UPB-PMRust/questions #16: Working with Shift Registers and LEDs

kevinvlad03, 2024-05-11:
# Question
I am trying to work with shift registers for my 36 leds to be on one after another in a sequence. I found a very good tutorial that encapsulates both hardware and software, from which I build the hardware part and also transformed the software part from MicroPy to my Rust Code. My code seems to be fine, hardware as well (*checked a lot of times*) but the leds seem to never turn on. I also checked hardware functionality, they are not burnt or anything like that. 
## Material read
You can check my code [here](https://gist.github.com/kevinvlad03/15c19e6d4ac3a697e15f3d81f8d0183a) and also the inspiration [here](https://peppe8o.com/how-to-use-74hc595-shift-register-with-raspberry-pi-pico-and-micropython/). Thank you!

### Later Update:
I have installed and tried the same code logic with MicroPy on Thonny, and all the hardware works perfectly, so the issue is with my code.

alexandruradovici, 2024-05-11:
I see that you have some debug messages, do you see them? I am trying to figure out if the software panics or not.

kevinvlad03, 2024-05-11:
Yes I am seeing them and the output seems to be the right one, at least close to it, I don t really know if that is exactly how it is supposed to be.

alexandruradovici, 2024-05-11:
Please put some `info!` statements in the `loop` and see that they continuously print. You can comment out the `info!` lines from the `shift_update`.

Did you run the python code using the exact same hardware setup?

kevinvlad03, 2024-05-12:
Hello! Sorry for the late update. So now there is a weird situation, there is still the same print pattern in the terminal after moving the 'info!' statment in the loop, but the very weird thing is that now 2 leds are up and running as they should, namely the last 2 leds on the shift register, pin 6 and pin 7. The rest are not working, so I guess there must be some kind of code missjudgement or bad logic? For the other question, yes I tested on the exact same hardware setup. I am waiting for other suggestions but now that I see at least something working I will try more. Thank you!

kevinvlad03, 2024-05-12:
Update: somehow now it works, I didn t modify anything and overnight some miracle happened I guess :). If you don t mind I will leave this question open if something happens when I continue connecting to the other shift registers. Thank you for the help!