UPB-PMRust/questions #17: [project] Path too long for compiling

cristianaprecup, 2024-05-11:
## Question
When I'm trying to cargo build i get this error that tells me that my compiling path is too long. I tried to move it on C:/ drive so that I shorten the path, and also to enable long path support in Windows but I didn't succeed to build it. What should I do?

[image]

## Details
I got this error after installing the crate for controlling servo motors: https://crates.io/crates/esp-idf-servo

cristianaprecup, 2024-05-11:
I managed to add my project in a Linux environment and I got the following error, this crate is not compatible with thumbv6m-none-eabi:  
[image]
I couldn't find any other useful crate for controlling the SG90 micro servo motor that I am using, do you know any? Thank you.

alexandruradovici, 2024-05-12:
You cannot use the ESP crates as ESP is a RISC-V and not ARM processor. It also uses the ESP-IDF library which is not available on the Raspberry Pici.

The servo should be controlled by a simole PWM which you can actually write.

Here you have an example for the microbit https://alelouis.eu/blog/pwm-microbit/. The API is a little different, but take into account just the PWM parameters and adapt them for your code.

This https://github.com/paunstefan/sg90_pi is a crate for the big Raspberry Pi, you can adapt the code for the Pico by exchanging tge used PWM API with the Pico's embassy PWM API.

alexandruradovici, 2024-05-23:
Is this solved?

cristianaprecup, 2024-05-23:
Yes, I managed to solve it.

closed 2024-05-23 by cristianaprecup