UPB-PMRust/questions #18: Problem with libraries for project

Angery-Stoman, 2024-05-23:
Rust is giving me this error because of a library in embassy:
can't find crate for `std`
the `thumbv6m-none-eabi` target may not support the standard library
`std` is required by `embassy` because it does not declare `#![no_std]`

alexandruradovici, 2024-05-23:
The embassy crate is `no_std`, can you please be more specific with the error? A screenshot maybe.

Angery-Stoman, 2024-05-23:
I tried using cargo build again and the error completely changed.
My lib.rs is currently empty.
[image]
These are the libraries that I am currently using and where the problem seems to stem from.
[image]

Angery-Stoman, 2024-05-23:
furthermore, this is my cargo.toml
[image]

alexandruradovici, 2024-05-23:
Please put here a link to the project repository.

Angery-Stoman, 2024-05-25:
It seems that I have solved the problem with the errors that I was getting. I just had a faulty settings.json and build.rs file.
I will close the issue.

closed 2024-05-25 by Angery-Stoman