UPB-PMRust/questions #19: [project] Pressure compensation formula

U-alb, 2024-05-23:
# Is my pressure calculation formula correct?

## Question
Hello! 
I need to measure the atmospheric pressure using the BMP280 in my project. I've tried to convert the C method found in the [documentation](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bmp280-ds001.pdf#page=45) to Rust, but I'm fairly sure that I've done something wrong, because it seems to break the execution of the code (i.e. panic)

A review would be greatly appreciated! Thank you
## Details
- Link to the code in question: https://github.com/UPB-FILS-MA/project-UdrAlberto/blob/20b508e8d2d84e6da99be3c7225e3a811df7021c/project/src/main.rs#L65-L90
## Material read
[Table that details the constants (of the format dig_Px) used in the C method](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bmp280-ds001.pdf#page=23)

U-alb, 2024-05-26:
Thanks for the suggestion!

It seems that the panic is caused by line 77: `let mut p: u32 = (1048576 - pressure_adc - ((var2 >> 12) as u32) * 3125) as u32;`

alexandruradovici, 2024-05-26:
In debug mode rust panics on:
- div by 0
- overflow
- underflow

To avoid this, you can use the `u32.saturating_add` and `u32.saturating_sub` functions or use `u64` numbers and, for bith, check for div 0.

Another solution, but with computation errors, is to build the software in release mode.

U-alb, 2024-05-26:
Everything works well, now. Thanks!

closed 2024-05-26 by U-alb