UPB-PMRust/questions #23: Rust error

alecsandraiordache, 2024-05-25:
# Question
I have an error regaring to the servo motor that i cannot fix. I tried to use also the labs.
## Material read
Link of my project repository : https://github.com/UPB-FILS-MA/project-alecsandraiordache

alexandruradovici, 2024-05-26:
Please describe what you did and what the error seems to be.

alecsandraiordache, 2024-05-26:
> Please describe what you did and what the error seems to be.
I am doing an automatic plant watering where use a servo motor that opens the bottle cap and I have these errors:
error[E0599]: no method named `set_duty` found for struct `Pwm` in the current scope
  --> src\main.rs:56:17
   |
56 |             pwm.set_duty(duty_cycle_value); // Set the duty cycle value
   |                 ^^^^^^^^ method not found in `Pwm<'_>`

error[E0599]: no method named `set_duty` found for struct `Pwm` in the curren

error[E0599]: no method named `set_duty` found for struct `Pwm` in the current scope
  --> src\main.rs:63:17
   |
63 |             pwm.set_duty(duty_cycle_value); // Set the duty cycle value 
   |                 ^^^^^^^^ method not found in `Pwm<'_>`

For more information about this error, try `rustc --explain E0599`.
error: could not compile `SoftwareMilestone` (bin "SoftwareMilestone") due to 2 previous errors

alexandruradovici, 2024-05-26:
The PWM driver dors not have that method, please tale a look at PWM lab for the syntax.

closed 2025-03-14 by alexandruradovici