UPB-PMRust/questions #25: [theory] &* usage in Rust

demesup, 2025-03-13:
During the course, we discussed the following code snippet(at the end of the description), and it was mentioned that if we had questions about the &* usage, we were free to ask. I wanted to clarify whether it’s simply used for deref-ref to obtain a &str type with its functionalities, or if there’s more to it?

## Links to materials that you read
https://stackoverflow.com/questions/41273041/what-does-combined-together-do-in-rust
- link to lecture:
https://pmrust.pages.upb.ro/docs/fils_en/lecture/02

Here’s the code snippet from the lecture for reference:
```
const SYS_CTRL_ADDR: usize = 0xe000_0000;

register_bitfields! {u32,
    AIRCR [
        VECTKEY OFFSET(16) NUMBITS(8) [KEY = 0x05fa],
        ENDIANESS OFFSET(15) NUMBITS(1) [],
        SYSRESETREQ OFFSET(2) NUMBITS(1) [],
        VECTCLRACTIVE OFFSET(1) NUMBITS(1) []
    ]
}

register_structs! {
    SysCtrl {
        (0xed0c => aircr: ReadWrite<u32, AIRCR::Register>),
    }
}

let sys_ctrl = unsafe { &*(SYS_CTRL_ADDR as *const SysCtrl) }; // C: struct SysCtrl *sys_ctrl = SYS_CTRL_ADDR;

sys_ctrl.aircr
    .modify(AIRCR::VECTKEY::KEY + AIRCR::SYSRESETREQ::SET);

```

alexandruradovici, 2025-03-14:
Let's break down the statement into several parts:

- `SYS_CTRL_ADDR` is of type `usize`, that is just a number
- `SYS_CTRL_ADDR as *const SysCtrl` casts the `usize` to a raw pointer to  `SysCtrl` that is stored at the `SYS_CTRL_ADDR` memory address

Writing 
```rust
let scr = SYS_CTRL_ADDR as *const SysCtrl;
```

is fine, but any method call of `scr` has to be wrapped into an `unsafe` block.

```rust
unsafe {
  sys_ctrl.aircr
      .modify(AIRCR::VECTKEY::KEY + AIRCR::SYSRESETREQ::SET);
  // the correct writing should be (*scr).aircr ..., but this is not required due to 
  // the fact that Rust usually dereferences automatically
}
```

To avoid this, we want to have a *safe* reference to `SysCtrl` instead of a raw pointer.

- `*(SYS_CTRL_ADDR as *const SysCtrl)` - technically represents the value that is stored at address `SYS_CTRL_ADDR`, in other word dereferences the raw pointer - this is the `unsafe` op, Rust forbids any raw pointer dereference in safe code, as the compiler does not know if there is anything valid stored at that address. We are not doing anything so far with the value, so it is not copied or moved from the memory address.
- `&` creates a *safe* reference to a value
-  `&*(SYS_CTRL_ADDR as *const SysCtrl)` - creates a *safe* reference to a the `SysCtrl` value, that is `&SysCtrl`. This reference can be used in safe code.

The `unsafe` should technically only surround `unsafe {*(SYS_CTRL_ADDR as *const SysCtrl)}` and `&` should be outside, but `unsafe` defines a block of code, an that implies that the `SysCtrl` value within the `unsafe` block is copied or moved outside of the `unsafe` block and placed in to the current scope (the `main` function in our case). `&` takes a reference to the moved value.

We cannot copy or move the value from the `SYS_CTRL_ADDR `, as that is the place where we have to write to the registers. If we copy or move it, we will write into a RAM location and not into the registers. This is why we need to surround everything in an `unsafe` block.

demesup, 2025-03-14:
Thank you for the explanation. I have no further questions for now.

closed 2025-03-14 by demesup

alexandruradovici, 2025-03-14:
I just read the StackOverflow link. In that case, `String` overrides the `*` operator. In our case, the data type is `* const SysCtrl` that does not override it.