UPB-PMRust/questions #26: [project] Unable to Flash Raspberry Pi Pico W 2

andronedrei, 2025-03-20:
## Question
My laptop is set up for compiling and flashing to the Raspberry Pi Pico W 2 (as tested successfully during lab), but I am having issues doing so at home with my own Pico W 2.

##Details
I'm using a recently bought Pico W 2 and I'm not sure if any prior setup is required. The board is recognized by my laptop and mounts successfully (showing the "index.htm" and "info_uf2.txt" files).  When executing 'cargo run', I get the following error:

Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.28s  
Running `probe-rs run --chip RP235x target/thumbv8m.main-none-eabihf/debug/lab-2025`  
Error: No connected probes were found.

alexandruradovici, 2025-03-20:
You have correctly described the problem, you need a probe. Another Pico can be used as a probe, please take a look at [Lab 02 - Debug section Secondary Pico](https://pmrust.pages.upb.ro/docs/fils_en/lab/01#secondary-pico) about how to wire and flash them.

alexandruradovici, 2025-03-20:
The Raspberry Pi Pico [Documentation](https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#debugging-using-another-pico-series-device) explains how to download the required firmware for the second Pico used as a debugger.

closed 2025-05-09 by alexandruradovici