UPB-PMRust/questions #27: [hardware] MLX90614ESF Infrared Temperature Sensor Wrong Readings

frihetselsker, 2025-04-12:
# Question
I have problems with an IR-sensor for body temperature measurement. It works, but the data sent is not relevant. For example, the temperature in my room is 24 °C. The sensor shows: 
12.035834 INFO  Ambient: 26.890015°C
12.036020 INFO  Object: 25.589996°C
 
I used both manual reading and a particular [crate](https://crates.io/crates/mlx9061x/0.3.0). Nothing changes. I also tried it on Arduino Uno, I got the same results. It was supposed to read body temperature, but the maximum value it read was only 29 degrees and it happened only when I touched the lamp. I tried with both 3.3V and 5V, no changes at all. What can I do with it?

## Hardware used
MLX90614ESF Infrared Temperature Sensor ([link](https://www.optimusdigital.ro/en/temperature-sensors/138-mlx90614esf-infrared-temperature-sensor.html?gad_source=1&gclid=Cj0KCQjwytS-BhCKARIsAMGJyzoTY9i3YbKmE_c_PXfKsUA3MDyLGtYE5YYbsu4E9VJfpMTTNrikHiUaAsQ1EALw_wcB))
# Materials read
[Datasheet](https://media.melexis.com/-/media/files/documents/datasheets/mlx90614-datasheet-melexis.pdf%23page29,)

alexandruradovici, 2025-05-09:
You might have to set up some parameters before you can read it. Look at the BMP280's code, you will see that before reading, it sets some parameters.

frihetselsker, 2025-05-09:
Actually it worked fine, the only thing I needed to do is just to process the obtained data a bit. Thank you!

closed 2025-05-09 by frihetselsker