UPB-PMRust/questions #28: [hardware] MAX30102 Heart Rate Sensor does not work

frihetselsker, 2025-04-12:
# Question
A heart rate sensor does not work at all. I couldn't find it using I2C scanning either on Raspberry Pi or Arduino Uno. For Raspberry Pi, it was just NACK, but for Arduino it just writes "Initializing..." and halts at this point. I read a bit over the Internet and found out there were some problems with MAX30100 on PCB. But I didn't find anything about MAX30102. What can I do with it?
## Hardware used
MAX30102 Heart Rate and Pulse Oximeter Sensor Module (Black) ([link](https://www.optimusdigital.ro/en/others/12530-max30102-heart-rate-and-pulse-oximeter-sensor-module-black.html?search_query=max&results=1794))
# Materials read
[Resolving the issue on MAX30100](https://electronicsinnovation.com/solved-max30100-not-working-initializing-pulse-oximeter-failed/)
[Arduino Forum](https://forum.arduino.cc/t/max30102-heart-rate-and-pulse-oximetry-sensor-not-working-led-light-not-turning-on-and-error-messages/1018339/8)

alexandruradovici, 2025-05-09:
Sorry, we missed this. I might be a faulty sensor or it might be under powered. Is it solved?

frihetselsker, 2025-05-09:
No problem! The sensor worked actually fine, I just needed to process the obtained data a bit. Thank you!

closed 2025-05-09 by frihetselsker