UPB-PMRust/questions #29: Error: No connected probes were found

AndreiRosu17, 2025-05-09:
# Question

Hello,

I made the wiring diagram for the board, and I'm going to put the debug code in (I held down the reset button while I plugged the wire into the laptop, let go of the button, added the file to the pico folder, and then that folder closed by itself).

In VSC, I wanted to run a lab skeleton, lab 3 to be exact, so that it would display Hello World in the terminal, but it wasn't to be because I got the error mentioned in the title.

The wires on the debug board are soldered, and I have headers on the compiler board. The wires were checked both before and after the application to see if a signal passes through them and there is no problem.

Sorry for bothering you

Have a good day!

## Hardware used
2x raspberry pi

[image]
[image]
[image]
[image]

AndreiRosu17, 2025-05-09:
when I insert the usb, it makes a sound as if I inserted a device into the system

alexandruradovici, 2025-05-09:
Did you connect both devices with USB cables to the PC?

alexandruradovici, 2025-05-09:
I think this might be the issue, both of them have to be powered.

AndreiRosu17, 2025-05-09:
The second one is powered. It is connected to the VCC second half breadboard, which is connected to the first one and it has power from the debug which is connected to USB. It is the same scheme 

[image]

alexandruradovici, 2025-05-09:
Can you run:

```
probe-rs list
cargo run
```

and paste the output here

AndreiRosu17, 2025-05-09:
Before this output, I have put the file on the debug storage, again, and after it closed itself. Then I opened the lab repo and wrote  next commands

[image]

alexandruradovici, 2025-05-10:
How and what firmware did you load to the debug Pico? 

Can you please run `ls usb` and place the output here?

AndreiRosu17, 2025-05-10:
I have downloaded the debugprobe_onpico2_uf2 from here https://github.com/raspberrypi/debugprobe/releases/tag/debugprobe-v2.2.2. With the microcontroller unplugged I was pressing the reset button until I plugged it in laptop, put the firmware in his storage and it has closed itself. 

This is the output when I run the lsusb output:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

alexandruradovici, 2025-05-12:
It does not seem to see your debug probe.

Is the debug pico 2 connected when you run `lsusb`?
Are you using Linux or WSL?

AndreiRosu17, 2025-05-12:
I resolved the problem, it's working when I am using the cargo run on a lab scheme. Thanks a lot for helping!

closed 2025-05-12 by AndreiRosu17

alexandruradovici, 2025-05-12:
Can you please write here what the problem was?

AndreiRosu17, 2025-05-12:
I didn't realize at first that a problem could be with the USB drivers, the laptop being new, but it couldn't read anything else except the floppy disk, but in the end I switched to working from Windows PowerShell directly because it reads without any problem, only in WSL/UBUNTU I had this problem.