UPB-PMRust/questions #30: SPI losing data during transfer

GameSkyRO, 2025-05-11:
Unaware of the root cause of the issue, I have created this issue as a blank one.
In my project, the first step of data acquisiton is sending the picture taken by the camera to the main controller, the STM32. In this case, the camera is the SPI slave, while the STM is the master. After analyzing the byte sequences sent and received, I see that some byte strings (16 bytes - the size I read each time) are missing in a random pattern. No matter the delay I create (from 5ms to 5000ms), byte strings are still missing.
I suspect that this has something to do either with:
- the STM32 not being able to pull the CS correctly, although the ESP does load new data into the buffer after each command
- the MISO line
- the spi transfer methods ( I have attempted to use both NoDMA and the DMA channels - no difference)

[image]
The index in front of each byte string represents the number of the transmission. Missing numbers indicate that nothing was read, but the loop was accessed with no errors.

alexandruradovici, 2025-05-12:
Do you have access to the ESP code? Can you log somehow every transmission that it does and compare it to the receiving logs?

GameSkyRO, 2025-05-12:
Yes. The esp code is visible in the repo. But to answer the question, yes, I am logging every tx buffer on every transfer. It sometimes happens that one buffer is sent from the esp side and the stm does not display it. This has happened even with 5s delay, as I watched the byte strings myself.

alexandruradovici, 2025-05-12:
Can you paste the link to your repo?

GameSkyRO, 2025-05-12:
[Repo](https://github.com/UPB-PMRust-Students/project-GameSkyRO/blob/main/)

alexandruradovici, 2025-05-12:
A few suggestions:
- printing buffers to UART / Debug takes times, the ESP might not be fast enough to send the next bit, try avoid to print the buffer on the ESP
- print any ESP errors and the number of the transmitted buffer, without the actual data and put the outputs (main and sub) here
- please read this https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/spi_slave.html#transaction-interval:
  - use queues 
    - wait for the frame request command
    - assign the whole buffer to the SPI queue
    - signal the main to start asking for transactions
  - use step by step signalling (slower)
    - the master asks for data
    - it waits for a GPIO (driven by the ESP) to change
    - ask for another transfer

I might be good to use the DMA sending the buffers.

alexandruradovici, 2025-05-14:
Any updates here?

GameSkyRO, 2025-05-14:
While I did not attempt all the given solutions, something weird is happening if the printing is not done on the esp side. The buffer seems to be going ahead, as in the esp shoots out the buffer faster and the stm receives a lot of 0s after half of the transmission. I am still testing different methods to understand the issues in output and will try to update with any other significant findings.
The queue approach also gives the same output with a lot of 0s.

GameSkyRO, 2025-05-15:
I have talked with the lab professor (Andrei Zamfir) and I will go forward with checking the memory via CubeProgrammer. I will return with further findings regarding overwriting or lack of memory space.

alexandruradovici, 2025-05-16:
I strongly suggest you synchronise the two transmissions using a GPIO pin.

GameSkyRO, 2025-05-16:
I have checked the memory of the stm and the issue appears on the access in memory. The image is fully sent and received from the esp to the stm with some bytes shifted by 1 bit, but that is less on an issue at the moment. I have tried printing the addresses of where my image is sent to the computer, and it clearly shows it skips some memory space. This is visible at 47c 57c, as it skips 4fc.

[image]

alexandruradovici, 2025-05-16:
Can you post the full output?

GameSkyRO, 2025-05-16:
```
0.011169 INFO  CS Pin initialized.
└─ sc::____embassy_main_task::{async_fn#0} @ src\bin\sc.rs:39
0.013397 INFO  SPI1 Initialized.
└─ sc::____embassy_main_task::{async_fn#0} @ src\bin\sc.rs:42
0.022491 INFO  image_data buffer start address: 0x200000f8
└─ sc::____embassy_main_task::{async_fn#0} @ src\bin\sc.rs:48
0.024871 INFO  Sending 0xA5 to ESP32 to request image capture...
└─ sc::____embassy_main_task::{async_fn#0} @ src\bin\sc.rs:56
0.331115 INFO  Requesting image size from ESP32...
└─ sc::____embassy_main_task::{async_fn#0}::{closure#0} @ src\bin\sc.rs:69
0.336822 INFO  Image size: 0 bytes
└─ sc::____embassy_main_task::{async_fn#0}::{closure#1} @ src\bin\sc.rs:78
0.338653 ERROR Invalid image size received
└─ sc::____embassy_main_task::{async_fn#0}::{closure#1} @ src\bin\sc.rs:81
1.340789 INFO  Sending 0xA5 to ESP32 to request image capture...
└─ sc::____embassy_main_task::{async_fn#0} @ src\bin\sc.rs:56
1.647308 INFO  Requesting image size from ESP32...
└─ sc::____embassy_main_task::{async_fn#0}::{closure#0} @ src\bin\sc.rs:69
1.653015 INFO  Image size: 0 bytes
└─ sc::____embassy_main_task::{async_fn#0}::{closure#1} @ src\bin\sc.rs:78
1.654846 ERROR Invalid image size received
└─ sc::____embassy_main_task::{async_fn#0}::{closure#1} @ src\bin\sc.rs:81
2.656982 INFO  Sending 0xA5 to ESP32 to request image capture...
└─ sc::____embassy_main_task::{async_fn#0} @ src\bin\sc.rs:56
2.969207 INFO  Image size: 2742 bytes
└─ sc::____embassy_main_task::{async_fn#0}::{closure#1} @ src\bin\sc.rs:78
2.971740 INFO  Starting image data transfer...
└─ sc::____embassy_main_task::{async_fn#0}::{closure#1} @ src\bin\sc.rs:99
1: 01 01 00 00 00 00 00 00 FF DB 00 43 00 0C 08 08
2: 0B 09 08 0C 0B 0A 0B 0E 0D 0C 0E 12 1E 14 12 10
3: 11 12 25 1A 1C 16 1E 2C 26 2E 2D 2B 26 2A 29 30
4: 36 45 3B 30 33 41 34 29 2A 3C 52 3D 41 47 4A 4C
5: 4E 4D 2F 3A 55 5B 54 4B 5A 45 4C 4D 4A FF DB 00
6: 43 01 0D 0E 0E 12 10 12 23 14 14 23 4A 32 2A 32
7: 4A 4A 4A 4A 4A 4A 4A 4A 4A 4A 4A 4A 4A 4A 4A 4A
8: 4A 4A 4A 4A 4A 4A 4A 4A 4A 4A 4A 4A 4A 4A 4A 4A
10: 4A 4A FF C4 00 1F 00 00 01 05 01 01 01 01 01 00
11: 00 00 00 00 00 00 00 00 01 02 03 04 05 06 07 08
12: 09 0A 0B FF C4 00 B5 10 00 02 01 03 03 02 04 02
13: 05 05 04 04 00 00 01 7D 01 02 03 00 04 11 05 12
14: 21 31 41 06 13 51 61 07 22 71 14 32 81 91 A1 08
15: 23 42 B1 C1 15 52 D1 F0 24 33 62 72 82 09 0A 16
16: 17 18 19 1A 25 26 27 28 29 2A 34 35 36 37 38 38
17: 3A 43 44 45 46 47 48 49 4A 53 54 55 56 57 58 58
19: 7A 83 84 85 86 87 88 89 8A 92 93 94 95 96 97 98
20: 99 9A A2 A3 A4 A5 A6 A7 A8 A9 AA B2 B3 B4 B5 B6
21: B7 B8 B9 BA C2 C3 C4 C5 C6 C7 C8 C9 CA D2 D3 D4
22: D5 D6 D7 D8 D9 DA E1 E2 E3 E4 E5 E6 E7 E8 E9 EA
23: F1 F2 F3 F4 F5 F6 F7 F8 F9 FA FF C4 00 1F 01 00
24: 03 01 01 01 01 01 01 01 01 01 00 00 00 00 00 00
25: 01 02 03 04 05 06 07 08 09 0A 0B FF C4 00 B5 10
26: 00 02 01 02 04 04 03 04 07 05 04 04 00 01 02 76
28: 13 22 32 81 08 14 42 91 A1 B1 C1 09 23 33 52 F0
29: 15 62 72 D1 0A 16 24 34 E1 25 F1 17 18 19 1A 26
30: 27 28 29 2A 35 36 37 38 39 3A 43 44 45 46 47 48
31: 49 4A 53 54 55 56 57 58 59 5A 63 64 65 66 67 68
32: 69 6A 73 74 75 76 77 78 79 7A 82 83 84 85 86 86
33: 88 89 8A 92 93 94 95 96 97 98 99 9A A2 A3 A4 A4
34: A6 A7 A8 A9 AA B2 B3 B4 B5 B6 B7 B8 B9 BA C2 C2
35: C4 C5 C6 C7 C8 C9 CA D2 D3 D4 D5 D6 D7 D8 D9 DA
37: F9 FA FF C0 00 11 08 00 78 00 A0 03 01 21 00 02
38: 11 01 03 11 01 FF DA 00 0C 03 01 00 02 11 03 10
39: 00 3F 00 CB FE CE BC 6F BB 6A 4F D5 D0 54 70 D8
40: EA 10 FC D7 56 FB 41 F4 74 38 AC 8E 4D 0B F6 B6
41: EF 32 1F 2F 19 F7 A8 CE 85 7A 49 E6 D3 FE FE B4
42: 2B 89 0E 1A 04 E7 3B A6 8D 4F B6 5A 9D FF 00 08
43: EB 7F CF F2 FF 00 E0 3D 05 87 FC 23 DE B7 4A 7E
44: ED 8D 20 F0 EC 7F C5 75 2F FC 05 54 50 17 2D 58
46: A3 6A 04 42 D5 0B 1A 62 35 34 EF F8 F4 4F A9 FE
[... 138 lines cut ...]
7.428802 INFO  Sending chunk from address: 0x200005fc
└─ sc::____embassy_main_task::{async_fn#0}::{closure#3} @ src\bin\sc.rs:175
7.454010 INFO  Sending chunk from address: 0x2000067c
└─ sc::____embassy_main_task::{async_fn#0}::{closure#3} @ src\bin\sc.rs:175
7.504394 INFO  Sending chunk from address: 0x2000077c
└─ sc::____embassy_main_task::{async_fn#0}::{closure#3} @ src\bin\sc.rs:175
7.529571 INFO  Sending chunk from address: 0x200007fc
└─ sc::____embassy_main_task::{async_fn#0}::{closure#3} @ src\bin\sc.rs:175
7.554748 INFO  Sending chunk from address: 0x2000087c
└─ sc::____embassy_main_task::{async_fn#0}::{closure#3} @ src\bin\sc.rs:175
7.579956 INFO  Sending chunk from address: 0x200008fc
└─ sc::____embassy_main_task::{async_fn#0}::{closure#3} @ src\bin\sc.rs:175
7.605133 INFO  Sending chunk from address: 0x2000097c
└─ sc::____embassy_main_task::{async_fn#0}::{closure#3} @ src\bin\sc.rs:175
7.630340 INFO  Sending chunk from address: 0x200009fc
└─ sc::____embassy_main_task::{async_fn#0}::{closure#3} @ src\bin\sc.rs:175
7.655517 INFO  Sending chunk from address: 0x20000a7c
└─ sc::____embassy_main_task::{async_fn#0}::{closure#3} @ src\bin\sc.rs:175
7.680694 INFO  Sending chunk from address: 0x20000afc
└─ sc::____embassy_main_task::{async_fn#0}::{closure#3} @ src\bin\sc.rs:175
7.705902 INFO  Sending chunk from address: 0x20000b7c
└─ sc::____embassy_main_task::{async_fn#0}::{closure#3} @ src\bin\sc.rs:175
7.718505 INFO  JPEG sent successfully
└─ sc::____embassy_main_task::{async_fn#0}::{closure#5} @ src\bin\sc.rs:186
```

GameSkyRO, 2025-05-16:
https://www.diffchecker.com/UqN0EoXv/

This is the last comparison I can think of to debug my program. SPI proves itself to be extremely unreliable, shifting bits, missing entire byte streams and being overall unable to achieve its goal. 
Maybe the sync pin could solve the missing bytes (I have not tried that yet, as it seems ridiculous that a communication protocol which already relies on a pin to be informed of when to send could require another check), but the bit shifting is confusing and does not point to any particular problem.
Even the fact that UART, via a cheap USB-TTL debugger, beats SPI is beyond me.
I will either try to implement the sync pin or move back to uart between the two boards, but that also seems to come with issues regarding overrunning the buffer.

GameSkyRO, 2025-05-17:
The sync pin made it worse, as there seems to be an even worse mismatch between the the byte streams sent and received.

GameSkyRO, 2025-05-19:
Issue fixed:
The sync pin was kept, as it informs the STM when the ESP has filled the buffer and is ready to send.
The greater issue regarding the missing bytes was on the ESP side, as there was not enough load time on it to fully fill the buffer. Added a delay here.
Solved the shifted bits by changing the SPI mode to 0 from 3.

closed 2025-05-19 by GameSkyRO