UPB-PMRust/questions #31: "Device not powered" error received when uploading code

annfkt, 2025-05-12:
# Question
I tried to upload the code to the board I'm using for the project (NUCLEO STM32F767ZI) and I get this error.

[image]

 The first time I tested Embassy on the board it worked with an example, but for a few days it hasn't worked anymore.
I tried connecting the board with different cables, and as a last resort even tested it from a different laptop but the same outcome occured each time.
At some piont I connected both usb ports which my board has to the laptop, trying to provide it with a bigger voltage, but nonetheless the same error appeard.
I am not quite sure where the problem is coming from or how to fix it, even tho I believe the board is causing me all this trouble.
## Hardware used
NUCLEO STM32F767ZI 
# Materials read

alexandruradovici, 2025-05-12:
The STM boards have sometimes a problem with stopping into debug mode. Try the following:
- press and hold the reset button (probably the the black one)
- while holding the reset button pressed, run the `cargo run` command
- as soon as it should start flashing, release the reset button

Try to repeat these steps a few times, it should eventually start flashing the board.

annfkt, 2025-05-13:
Sorry, I accidentally missclicked when trying to add a comment on the results which i received as i further tried to resolve the prblem. Please reopen the issue.

As I wanted to

annfkt, 2025-05-13:
As I wanted to mention, i have tried the sollution that you mentioned once before creating the issue and also a few times after you  offeredd it as a solution but still I received the same error.

alexandruradovici, 2025-05-14:
Please post an image with you device and the connection to the laptop.

alexandruradovici, 2025-05-14:
Please run `probe-rs list` and post the output here.

alexandruradovici, 2025-05-15:
I am assuming the error is the serial port that has no driver.

Please try the press and release of the reset button several times (10x +). It should work.

alexandruradovici, 2025-05-15:
Please open the `Device Manager` and look for devices with an exclamation mark next to them and place a screenshot here.

annfkt, 2025-05-16:
[image]

alexandruradovici, 2025-05-16:
Please open `Universal Serial Bus Devices`.

Did you try several times to push and release the reset button?

annfkt, 2025-05-16:
Yes, I have tried it multiple times and still I have seen no changes when connecting my board.