UPB-PMRust/questions #32: Problem with SD card in project

rusucarla, 2025-05-16:
I have an audio player for the project.

The idea was reading in real time from the SD card, converting 16bit to 32bit (stereo) in the Pico and then sending the data to a DAC.

Unfortunately, the SD card is extremely unreliable. I'm using the [embedded-sdmmc](https://crates.io/crates/embedded-sdmmc) crate.

I tried to implement the 80 clocks at 400 kHz before upping the speed. I tried to do health cheks and testing prior to reading the file. 

The errors I run into are:
- Seeing the card size as much bigger than its actual size (64 -> 1.6 tb)
- Opening the card, seeing the type and size, but not able to open the volume manager
- Opening the volume manager but not able to read the header of .WAV files
- Managing the above but not being able to reopen the files when trying to actually playback the files
- When running into a reading error, not being able to "revive" the card

alexandruradovici, 2025-05-16:
Can you please post a link to your repo so we can talk a look?

@NegrilaRares you used FAT in your project last year, can you chip in on this?

alexandruradovici, 2025-05-16:
This is @NegrilaRares's project https://pmrust.pages.upb.ro/docs/fils_en/project/2024/rares.negrila where he explains the issues that he had https://pmrust.pages.upb.ro/docs/fils_en/project/2024/rares.negrila#week-6---12-may.

rusucarla, 2025-05-16:
This is the [repo](https://github.com/UPB-PMRust-Students/proiect-rusucarla).

This is the fdisk output: 
```
Disk /dev/sdb: 59,52 GiB, 63910707200 bytes, 124825600 sectors
Disk model: Mass-Storage    
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x096182c2

Device     Boot Start       End   Sectors  Size Id Type
/dev/sdb1       32768 124825599 124792832 59,5G  c W95 FAT32 (LBA)
```

I think it's ok as it fits the parameters of the [example code](https://github.com/rp-rs/rp-hal-boards/blob/main/boards/rp-pico/examples/pico_spi_sd_card.rs) given by the crate git (lines 35-36):
```
//! The important bit here is the _Type_ with `W95 FAT32 (LBA)`, other types
//! are rejected by the `embedded_sdmmc` filesystem implementation.
```

The difference between my project and Rares's one is that I'm using the SpiDeviceWithConfig (recommended in the SPI lab) to commute safely between the LCD and the SD card.

I could try using both SPI channels and seeing if the problem is the LCD but the weird part is that sometimes the SD card behaves weirdly even when the code does not initialize the LCD (see the /src/bin/test_sd.rs in the repo).

I could also try to use just SpiDevice and both SPI channels to see if that's where the problem was ?

alexandruradovici, 2025-05-16:
Are you using the sync or async API?

Try using the sd card on a different SPI bus that has no other peripherals connected.

rusucarla, 2025-05-16:
I am using the `sync` API like in the lab:

> For this, we can use an Embassy driver called SpiDevice, which will allow us to connect both the screen and the sensor in the same SPI configuration! Since the screen [is already using a blocking SPI](https://pmrust.pages.upb.ro/docs/acs_cc/lab/05#screen), we will use it when creating our SPI sensor device

I will try to separate the SPI devices and see if that works. If I separate them can I use `async` spi ? Or do they both need a `sync` spi ?

NegrilaRares, 2025-05-16:
The SD formating looks fine.

I personally used a separate SD card reader, so I am not too sure how the one integrated with the LCD would interact with the library. (Here's the one I used: [[SD Card Reader](www.optimusdigital.ro/en/others/98-sd-card-reader.html)])

The link to your repo doesn't open for me so I cant exactly pinpoint a direct problem right now.

rusucarla, 2025-05-16:
About the repository, it's private and I can't fork it nor can I change the visibility. I did create a public [clone](https://github.com/rusucarla/proiect-rusucarla-clone).

Unfortunately Optimus Digital does not have the SD card reader anymore. I'll try to work with what I have over the weekend and if it doesn't work I'll buy a separate reader and maybe a new card ? Did you have many problems with the microSD in a SD cover that were resolved by switching to an actual sd card ?

NegrilaRares, 2025-05-16:
For the final version of the project i used a microSD in an SD case so i would say it should work, but by that point the code was working with normal SD cards and with microsd readers.

alexandruradovici, 2025-05-16:
@NegrilaRares I added you to the project repository.

rusucarla, 2025-05-16:
I just completed moving the SD to SPI1 and it seems to be working ? At least for now 😆 (i did push the code but it's in a patch fix phase for now)

Thank you for all the help. I will come back over the weekend if it starts having issues again (but hopefully it will not).